PiSCSI/piscsi #1418: Unit test assertions failure in ScsiCdTest_ModeSelect_Test::TestBody

rdmark, 2024-01-16:
When run in debug mode, there is an assertion failure in `ScsiCdTest_ModeSelect_Test::TestBody`. This is only triggered when unit tests are compiled with the `DEBUG=1` flag.

Uwe spotted and reported this in https://github.com/PiSCSI/piscsi/issues/1415 -- see that ticket for the stack trace with symbols.

rdmark, 2024-01-16:
@kkaempf I think this failed assertion was introduced with https://github.com/PiSCSI/piscsi/pull/1406

Would you be able to have a look?

FWIW I'm proposing we enable DEBUG for unit tests in the CI workflow as well, to catch these earlier in the future: https://github.com/PiSCSI/piscsi/pull/1417

kkaempf, 2024-01-16:
I can't reproduce a test crash on my [issue-1410](https://github.com/kkaempf/piscsi/tree/issue-1410) branch (forked from ad5eae93e) when compiled with `DEBUG=1` 🤷🏻‍♂️

kkaempf, 2024-01-16:
I'll rebase and test again ...

kkaempf, 2024-01-16:
Rebased onto 7a986c84aa54fa550e1255f5ff83174b294fa3f4 I get a crash

PR #1421 (2024-01-18): Ensure DiskCache constructor is called with a blocks count > 0
otherwise its constructor assertion will fail when compile with DEBUG=1

Fixes #1418
commit 7cc8df27 (2024-01-18): DiskCache needs a size

otherwise its constructor assertion (block > 0) will fail.

Fixes #1418
closed 2024-01-19 by rdmark