PiSCSI/piscsi #1419: emu3fs (Emu III and IV Samplers) implementation within PISCSI

alhargrave, 2024-01-16:
# Info

- Which version of Pi are you using: rPi3
- Which github revision of software: latest
- Which board version: FULL
- Which computer is the PiSCSI connected to: Emu ESI/32 Sampler
- Which OS you are using (output of 'lsb_release -a'): raspian release 11 (bullseye)

# Describe the issue

I have wanted to get PISCSI (and before that RASCSI) to work with my old emu esi/32 sampler. Emu use a bespoke file system on CDs and in formating disks for use with the sampler. There is an excellent github project devoted to EIII and EIV formats within linux. I have implemented within the PISCSI and tested by mounting an iso image of one of the sampler CDS. I can read the list of samples within the iso from the CLI of the PISCSI. Currently i cannot access from the sampler itself (the PISCSI is attached on a centronics cable externally on the SCSI port. 

My ask is
1. How do i connect the iso to the sampler - are there other examples i could attempt to follow or a method i am missing?
2. Can the drivers be implemented formally within PISCSI with workflow using the web interface

The link to the emu3fs project is as here https://github.com/dagargo/emu3fs#readme

Thank you

rdmark, 2024-01-16:
Have you read the documentation at https://github.com/PiSCSI/piscsi/wiki/Drive-Setup ? It describes how you attach an image file to expose it to your scsi host system. 

About your second question: is this about copying files from the Linux file system onto the file system of an image file?

alhargrave, 2024-01-16:
i tried mounting the iso as stated in the documentation and as a CD-ROM device and the sampler could see it but not read it as lacking emu filesystem. hence my question. i worked on getting the the filesystem - now how do i mount it so the sampler can see it?

My second question was about expanding the driver capabilities of the PISCSI with inclusion of the work donw on emu3fs (or any other bespoke filesystem that might be out there - this could also be a method for users adding the formats themselves

alhargrave, 2024-01-16:
it could be that instead of mounting the iso as cd i do that as a hard drive. Within the emu3fs docs

Mounting ISO images
ISO images can be accessed through loop devices. In this example, we are using the loop0 device.

$ sudo losetup /dev/loop0 image

they use loop devices. And to answer the question on corruption:

pi@piscsi:/mnt/Vol1 $ sudo modprobe emu3_fs
pi@piscsi:/mnt/Vol1 $ sudo mount /dev/loop0 /mnt/Vol1/
pi@piscsi:/mnt/Vol1 $ cd /mnt/Vol1/
pi@piscsi:/mnt/Vol1 $ ls -al
total 173830
drwxrwxrwx 2 root root    1536 Jan 16 15:23  .
drwxr-xr-x 3 root root    4096 Jan 10 17:26  ..
-rw-r--r-- 1 pi   pi   3320881 Jan 16 15:23 '12 String Guitar'
-rw-r--r-- 1 pi   pi   3466239 Jan 16 15:23 '4 Piece Horns 4M'
-rw-r--r-- 1 pi   pi   6889423 Jan 16 15:23 '4 Piece Horns 8M'
-rw-r--r-- 1 pi   pi   4180421 Jan 16 15:23 '9ft Grand Piano'
-rw-r--r-- 1 pi   pi   3191597 Jan 16 15:23 'Brass Bank'
-rw-r--r-- 1 pi   pi   3961141 Jan 16 15:23 'Cymbal Selection'
-rw-r--r-- 1 pi   pi   4164231 Jan 16 15:23 'Dance Club'
-rw-r--r-- 1 pi   pi     65536 Jan 16 15:23 'E3 Main Code'
-rw-r--r-- 1 pi   pi      1867 Jan 16 15:23 'E-mu Banks 1-44'
-rw-r--r-- 1 pi   pi   3627275 Jan 16 15:23 'Flautas Bonita'
-rw-r--r-- 1 pi   pi   3867137 Jan 16 15:23 'Full Arco String'
-rw-r--r-- 1 pi   pi   4005625 Jan 16 15:23 "Gong 'O' Rama"
-rw-r--r-- 1 pi   pi   4094557 Jan 16 15:23 'Groupo Sinfonia'
-rw-r--r-- 1 pi   pi   3566719 Jan 16 15:23 'Harp&FemaleChoir'
-rw-r--r-- 1 pi   pi   2208811 Jan 16 15:23 'Johnny Bassman'
-rw-r--r-- 1 pi   pi   2997101 Jan 16 15:23 'Kyodai Bang'
-rw-r--r-- 1 pi   pi   2887143 Jan 16 15:23 'LA Synth 1'
-rw-r--r-- 1 pi   pi   4121107 Jan 16 15:23 'Latin Hand Perc'
-rw-r--r-- 1 pi   pi   3816547 Jan 16 15:23 'MultiBass Select'
-rw-r--r-- 1 pi   pi   4197155 Jan 16 15:23  OrchestralComp#1
-rw-r--r-- 1 pi   pi   4174509 Jan 16 15:23 'Pipe Organ 4Meg'
-rw-r--r-- 1 pi   pi   3980463 Jan 16 15:23 'Pop Ahs'
-rw-r--r-- 1 pi   pi   4015425 Jan 16 15:23 'Pop Oohs'
-rw-r--r-- 1 pi   pi   4091853 Jan 16 15:23 'Rock Party,Dudes'
-rw-r--r-- 1 pi   pi   4178353 Jan 16 15:23  SecViolinTrils4M
-rw-r--r-- 1 pi   pi   8332373 Jan 16 15:23  SecViolinTrils8M
-rw-r--r-- 1 pi   pi   4077843 Jan 16 15:23 'Solo Trombone 4M'
-rw-r--r-- 1 pi   pi   3161865 Jan 16 15:23 'Solo Trumpet 4M'
-rw-r--r-- 1 pi   pi   6270921 Jan 16 15:23 'Solo Trumpet 8M'
-rw-r--r-- 1 pi   pi   2842827 Jan 16 15:23 'Solo Violin'
-rw-r--r-- 1 pi   pi   4065433 Jan 16 15:23 'Soul Ahs'
-rw-r--r-- 1 pi   pi   3961139 Jan 16 15:23 'Soul Oohs'
-rw-r--r-- 1 pi   pi   4062815 Jan 16 15:23 'Sound FX'
-rw-r--r-- 1 pi   pi   4180483 Jan 16 15:23 'Stereo Cymbals'
-rw-r--r-- 1 pi   pi   3160815 Jan 16 15:23 'Stereo F Horns'
-rw-r--r-- 1 pi   pi   3148455 Jan 16 15:23  StereoGrandPiano
-rw-r--r-- 1 pi   pi   4188259 Jan 16 15:23  StereoSteelDrums
-rw-r--r-- 1 pi   pi   3251117 Jan 16 15:23 'Stereo Strings'
-rw-r--r-- 1 pi   pi   3264327 Jan 16 15:23 'Ster Synth Combo'
-rw-r--r-- 1 pi   pi   2641667 Jan 16 15:23 'Str Ac. Guitar'
-rw-r--r-- 1 pi   pi   3909589 Jan 16 15:23 'Tenor Sax'
-rw-r--r-- 1 pi   pi   3560611 Jan 16 15:23 'Textural Strings'
-rw-r--r-- 1 pi   pi   3749339 Jan 16 15:23 "Verb'd Out Drums"
-rw-r--r-- 1 pi   pi   2542187 Jan 16 15:23  VibesMarimbaPerc
-rw-r--r-- 1 pi   pi   3941497 Jan 16 15:23 'Vintage Synths'
-rw-r--r-- 1 pi   pi   4204147 Jan 16 15:23 "World O' Synth 1"

Thats the sampler bank for Vol1 of the emu es/32 sampler series. so no corruption when running from PISCSI CLI

rdmark, 2024-01-16:
I know that certain samplers expect you to mount CD-ROM images has hard drives in piscsi, f e by changing the file ending. This is something you can try.

If you haven’t done so already, please go through each step in the troubleshooting guide:
https://github.com/PiSCSI/piscsi/wiki/Troubleshooting — it can be something as simple as termination being incorrectly set, or a faulty scsi cable (both common issues.)

rdmark, 2024-01-30:
Closing this ticket. The first question has been addressed, and regarding the second request, it is not within the scope of piscsi to mount file systems contained on images. While I can see the utility, there is just too broad of a scope (FAT, HFS, historical *NIX file systems, proprietary sampler file systems, etc.) and too much UI to create around it.

closed 2024-01-30 by rdmark