PiSCSI/piscsi #1427: MODE SELECT ignores mode pages in develop branch, v23.11.01 works

uweseimet, 2024-01-31:
This ticket refers to commit a23d642 (current develop branch).

When you send a sequence of mode pages with MODE SELECT(6), some pages are ignored. If one of the pages is a format device page for a sector size change (configured in the page data, not in the descriptor), and when this page is located somewhere at the end of the list, the change is ignored. The log entry about the sector size change does not appear. When you move this page up in the page list the log entry appears. Whether a page is evaluated or not seems to depend on its position in the list of mode pages.

With release v23.11.01 this is correct, i.e. for the same command parameters the sector size change request is logged (for both MODE SELECT(6) and MODE SELECT(10)), regardless of the size of the page list.

rdmark, 2024-02-17:
Thanks for reporting, Uwe.

@kkaempf At a glance, this seems to be a specific scenario that we didn't consider when reviewing your ModeSelect PR series.

https://github.com/PiSCSI/piscsi/pull/1405
https://github.com/PiSCSI/piscsi/pull/1406

rdmark, 2024-03-20:
@uweseimet Does the unit test added here adequately represent the test case that you describe in this issue? https://github.com/PiSCSI/piscsi/pull/1443

uweseimet, 2024-03-20:
@rdmark No, it doesn't.. The test needs to send *multiple* pages (a list of pages) with a *single* MODE SELECT. Then, depending on where in the list the sector size is changed, the change does not have any effect. You can see this in the logfile: piscsi in the develop branch does *not* log that the sector size has been changed, v23.11.01 does. This shows that the code parsing the page list passed with MODE SELECT is broken in the develop branch.

Please note that I do not intend to spend more time on this and the other MODE SELECT bug. I thought I'd report what I found, so that it does not end up in a release, but I am not interested in defending why I think that a program that crashes has a bug.

rdmark, 2024-03-21:
@uweseimet Can you please share a MODE SELECT(6) payload that you used to demonstrate the bug? This would help avoid making incorrect assumptions or mistakes based of a misunderstanding of SCSI behavior.

uweseimet, 2024-03-21:
The payload (list of pages) was similar to the one below. It may even have been the same, but I don't know anymore.
```
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
02:01:00
03:16:00:08:00:00:00:00:00:00:00:19:08:00:00:01:00:0b:00:14:00:00:00:00
```

uweseimet, 2024-03-24:
@rdmark Out of curiosity I diffed scsi_command_util.cpp against the last release and found this:
```
-               length -= size;
+               length -= size + 1;
                offset += size;
``` 
This looks quite wrong to me. Why is the length now decremented faster than the offset is incremented? IMHO this results in the loop being terminated too early, before all data have been parsed.

rdmark, 2024-03-26:
@uweseimet if you look at the git blame for that change you find this commit: https://github.com/PiSCSI/piscsi/commit/ad5eae93e7f721817c95a01285babdb97ed940b8

The commit message lays out the justification for the logic change in detail. This is the relevant passage. Do you have a different interpretation of this data?

```
* Fix page length computation in ModeSelect

tl;dr

The 'skip to next ModeSelect page' computation was off-by-one, either not
taking the page code itself into account or missing the fact that the
page length is given as `n - 1`.

Fix:

Add 1 to the computed length.

Details:

OpenVMS Alpha sends a ModeSelect6 as follows
~~~
command:

ModeSelect6, CDB $151000001900

payload:

 0  1  2  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24
00 00 00 08 00 00 00 00 00 00 02 00 01 0a 24 00 00 00 00 00 00 00 00 00 00
~~~

This translates to (accoring to [1], Section 8.3.3)

~~~
Mode Data Length 0
Medium Type      0
Device-specific  0
Block desc len   8
~~~

with the following offset / length computation _before_ the `while` loop
~~~
offset = 12
length = 13
~~~

The first payload section is
~~~
 4  5  6  7  8  9 10 11
00 00 00 00 00 00 02 00
~~~

translating to

~~~
Density Code     0
Number of blks   0
Reserved         0
Block length   0x200 512
~~~

Then follows a pagecode 1 as
~~~
12 13 14 15 16 17 18 19 20 21 22 23 24
01 0a 24 00 00 00 00 00 00 00 00 00 00
~~~

translating to
~~~~
Page code        1
Page length -1  10
Mode parameters 24 00 00 00 00 00 00 00 00 00 00
~~~

computing (inside the `while` loop, as `// Advance to the next page`)

~~~
size =  10 + 2 = 12
~~~

followed by new `offset` and `length` values

~~~
offset = 25
length = 1
~~~

So it stays in the `while` loop (and has a larger-than-buffer `offset`
value)
```

kkaempf, 2024-03-29:
> The payload (list of pages) was similar to the one below. It may even have been the same, but I don't know anymore.
> 
> ```
> 02:01:00
> ...
> 02:01:00
> 03:16:00:08:00:00:00:00:00:00:00:19:08:00:00:01:00:0b:00:14:00:00:00:00
> ```

Where is this data coming from ? 

I'd assume this would be "mode pages". Looking at the definition (Table 94 in [Small Computer System Interface - 2 rev 10L.pdf](https://dn790004.ca.archive.org/0/items/SCSISpecificationDocumentsSCSIDocuments/Small%20Computer%20System%20Interface%20-%202%20rev%2010L.pdf) - page 148 in document, 184 in pdf) it has

- page code (1 byte)
- page length **n - 1** (!)  (1 byte)
- page data (n bytes)

So the above should rather be
> ```
> 02:00:00
> ...
> 02:00:00
> 03:15:00:08:00:00:00:00:00:00:00:19:08:00:00:01:00:0b:00:14:00:00:00:00
> ```

Commit https://github.com/PiSCSI/piscsi/commit/ad5eae93e7f721817c95a01285babdb97ed940b8 was based on actual captured SCSI data with 2 pages.

I'm going to extend the testcase now to cover > 2 pages. It might well be that the offset / length computation still is wrong for > 2 pages.

uweseimet, 2024-03-29:
The specification says: "The page length field specifies the length in bytes of the mode parameters that follow. If the initiator does not set this value to the value that is returned for the page by the MODE SENSE command, the target shall terminate the command with CHECK CONDITION status."

When the page data are 02:01:00 there is 1 byte following the page length field, i.e. the page length field must be 1. Please look at the mode page examples in the specification, e.g. table 96. In table 96 are 6 bytes following the length field and the length field is 6. This is consistent with the mode page data I used in my tests. Correct me if I am wrong, but if you were right the page length in table 96 would be 5, which is not the case.

kkaempf, 2024-03-29:
>Please look at the mode page examples in the specification, e.g. table 96. In table 96 there are 6 bytes following the length field and the length field is 6. This is consistent with the mode page data I used in my tests. Correct me if I am wrong, but if you were right the page length in table 96 would have to be 5, which is not the case.

Looks like we have a conflict between an example in the spec (length byte == number of byte following) and the actual data captured from the DEC BIOS (length byte == number of bytes following _minus 1_) 😉 

Do we need a DEC BIOS flag ? 🤔 

Anyone with more captured data to look at ?

uweseimet, 2024-03-29:
@kkaempf Before you try to add any flag I recommend to verify that the DEC BIOS really violates the specification. I doubt it, because in this case it would not be compatible with any SCSI drive that correctly implements the specification. I bet the vast majority of drives do it right, because otherwise they would not work with most of the available drivers/tools. It is more likely that you have misinterpreted something.

Do you have any real SCSI drive that is compatible with your DEC BIOS? In this case I recommend that you verify the behavior of this drive regarding MODE SELECT. If you are right, in order to be compatible with the DEC BIOS the drive would have to violate the specification. You can use your PISCSI board and the s2pexec tool of https://www.scsi2pi.net/, for instance, to run any SCSI command against this drive and to evaluate the results.

If you are convinced that your BIOS is a special case I would be interested if https://www.scsi2pi.net/, which has a more elaborate MODE page handling than PiSCSI, works with your board or not. Testing this does not take much time, you can simply install an SCSI2Pi binary package.

In any case, what the specification says should be the default implementation.

@rdmark All in all, it's up to you, of course, but as you can see from my comments I doubt once again that any kind of "quirks" mode is needed. Real SCSI drives also do not have this kind of mode and work with all kinds of platforms/drivers. Usually in such a case there is just a misunderstanding of the specification or what the drivers do.

uweseimet, 2024-03-29:
@kkaempf I thought I'd run a test with your payload and the SCSI2Pi in_process_test (see https://github.com/uweseimet/scsi2pi/wiki/Advanced-Testing) myself, as a proof of concept. This is the result:
```
>./bin/in_process_test -s "-i 0 zip.hds"
s2pexec>-i 0 -c 151000001900 -d 000000080000000000000200010a2400000000000000000000
[17:52:57.144] [warning] Device reported CHECK CONDITION (status code $02)
Error: ILLEGAL REQUEST (Sense Key $05), INVALID FIELD IN PARAMETER LIST (ASC $26), ASCQ $00
>s2pexec
```
The error is expected because your payload appears to contain a vendor-specific mode page 0, which is not supported by the default configuration. As soon as I add a user-defined mode page 0 (see https://www.scsi2pi.net/de/properties.html) your payload is accepted:

```
>./bin/in_process_test -s "-i 0 zip.hds"
ss2pexec>-i 0 -c 151000001900 -d 000000080000000000000200010a2400000000000000000000
>s2pexec
```
This means that nothing is wrong with how PiSCSI v23.11.01 parses the mode pages. But PiSCSI does not support mode page 0, and this appears to cause the problem with your payload.

Note that for a valid solution PiSCSI has to support mode page 0 for both MODE SENSE and MODE SELECT.
User-configurable (on device level) mode page data are IMO the only valid solution, because each device that supports mode page 0 can have different mode page data. If you just implement support for mode page 0 as it is needed by your setup, as soon as the there is another user with a setup that needs a different mode page 0, you have a conflict: Without a user-definable mode page, why should PiSCSI support mode page 0 just like you need it, and not just like the other user(s) need it?

kkaempf, 2024-04-01:
@uweseimet, I stand corrected on the offset/length calculation in `scsi_command_util.cpp`. See https://github.com/PiSCSI/piscsi/pull/1447 for a fix.

However, I don't get your "page 0" arguments about the Alpha VMS payload.
`000000080000000000000200010a2400000000000000000000` splits up into
`00000008 0000000000000200 010a2400000000000000000000` as explained in https://github.com/PiSCSI/piscsi/issues/1427#issuecomment-2021621269

Looking at the `010a2400...` - it's page 0x01 request with 0x0a bytes following. However, 0x0b bytes actually follow.
The "trailing" byte 0 could be interpreted as a "page 0" but it's missing the length byte. 🤷🏻‍♂️

uweseimet, 2024-04-01:
@kkaempf Page 0 does not have/require a length byte. Please check the SCSI specification. This page does not have any well-defined format. This is why it must be the last page of a page list, because otherwise you would not be able to determine the start of the next page.

rdmark, 2024-04-04:
To recap, the reason we wanted to implement the DEC vendor specific ModeSelect page 0 was because the DEC machine cannot boot from a piscsi volume without it:

https://github.com/PiSCSI/piscsi/issues/1402

The question is if there’s another way to handle the vendor page, I e ignoring or retuning some kind of stub value that makes the DEC boot without introducing SCSI spec breaking behavior.

rdmark, 2024-04-13:
Whole changeset reverted with https://github.com/PiSCSI/piscsi/pull/1451

closed 2024-04-13 by rdmark