PiSCSI/piscsi #1434: Fix wrong warning about CD-ROM file size

uweseimet, 2024-02-19:
A minor issue, but not correct anyway: Due to a wrong comment in the legacy sources from a long time ago, piscsi reports (or does not report) a warning when it should not (or should).
```
if (size % 2536) {
```
must be
```
if (size % 2352) {
```

rdmark, 2024-02-19:
How does this calculation work, and what is 2352 derived from? I was under the impression that a correct file size would be divisible by 2048 (the sector size of an iso file system.)

uweseimet, 2024-02-19:
@rdmark 2352 includes checksum bytes etc. You can find details in the web.
IMHO this code is unlikely to be executed anyway, because as far as I can tell the rest of the code is wrong. I just pointed this out because it was me who changed it because of a misleading comment in the original code. The comment said 2536 bytes, and the code said 2352 bytes. The latter is correct.

rdmark, 2024-02-27:
For my own curiosity I looked at the initial commit and can confirm that the Japanese language code comment does not match the code: https://github.com/PiSCSI/piscsi/blob/35866cfb84ecb9657798a219f0572c97d9424c39/src/raspberrypi/disk.cpp#L3950

uweseimet, 2024-02-27:
Even if it had matched it would have been wrong ;-).

closed 2024-02-27 by uweseimet