PiSCSI/piscsi #1439: Macintosh Plus not Seeing Pi

Jeremiahdmass, 2024-03-01:
# Info

- Which version of Pi are you using: RPI 3 B v1.2
- Which GitHub revision of software: November 2023 release
- Which board version: PiScsi Zero v1.0 FULLSPEC
- Which computer is the PiSCSI connected to: Macintosh Plus
- Which OS you are using (output of 'lsb_release -a'): Raspbian GNU/Linux 11 (bullseye)

# Describe the issue
I can't get the Macintosh to recognize the PiScsi at all. I have tried several different disk images, none of which have worked. The Macintosh boots to the question mark no matter what I have attached on the PiScsi. I believe it may have something to do with the Plus's weird SCSI standards. 

I think it may have something to do with the Unit Attention, SCSI Selection Delay, and the SCSI selection pulses, because that has caused issues on the SCSI2SD. I have no idea how these settings could be adjusted on 
 the PiScsi though. [https://www.codesrc.com/mediawiki/index.php/Mac_Plus]

rdmark, 2024-03-01:
@Jeremiahdmass This sounds like either a SCSI termination or cablage issue. Please go through all the steps in https://github.com/PiSCSI/piscsi/wiki/Troubleshooting and see if it makes a difference.

FWIW there are known quirks and limitations with piscsi and the Mac Plus. See https://github.com/PiSCSI/piscsi/issues/656 for more info.

Jeremiahdmass, 2024-03-01:
Thanks for the reply. I put a glob of solder on the termination pads and have the db25 connector plugged directly into the Mac's SCSI port so I don't think either of those could be an issue. However, I forgot to mention that when I look in the logs, there's nothing there besides it saying that I mounted whatever I was trying to boot from.

rdmark, 2024-03-01:
Are other SCSI mass storage devices working properly with the same Plus and SCSI cablage?

Jeremiahdmass, 2024-03-02:
I tried some images formatted with the silver lining driver, and still nothing. I would think a driver issue would boot to a sad Mac instead of just waiting for a floppy.

rdmark, 2024-03-02:
Your symptoms make me think the signals aren’t getting through on the SCSI bus. Which suggests it’s either bad cablage, heavy oxidation on the DB25 ports, fault on the piscsi board or fault on the Plus logic board. 

What’d I would do first is to try other SCSI cables and other Macs to eliminate those factors. And also work some contact cleaner into those DB25s to clear out any oxidation.

Jeremiahdmass, 2024-03-02:
Unfortunately I don't have another Mac to try it on. I have worked some contact cleaner into the port but I guess I can try some more. I definitely am going to buy some more cables though.

Jeremiahdmass, 2024-03-12:
I have confirmed that my connections are good. The PiScsi is now showing some activity. When the Mac is searching for a SCSI device, the ENB light flashes on once. But the Mac still says it can't find any SCSI devices. I've tried several different hard drive configurations and disk images but to no avail.

rdmark, 2024-03-20:
Did you try the Plus specific Silverlining formatted image that I shared in the issue ticket linked above?

If that one doesn't work either, please share piscsi trace logs from when the Plus is powered on.

Jeremiahdmass, 2024-03-23:
I have tried a Silverlining formatted disk. However, there is nothing in the logs except the attached device even after the light blinks and the Mac can't find it. 
Here are the logs.
`Mar 23 23:25:52 piscsi PISCSI[495]: [2024-03-23 23:25:52.109] [info] Validating: operation=ATTACH, command params='locale=en', 'token=???', device=6:0, type=SCHD, device params='file=HD1-100MB_HFS.hda', vendor='', product='', revision='', block size=0
Mar 23 23:25:52 piscsi PISCSI[495]: [2024-03-23 23:25:52.110] [info] Executing: operation=ATTACH, command params='locale=en', 'token=???', device=6:0, type=SCHD, device params='file=HD1-100MB_HFS.hda', vendor='', product='', revision='', block size=0
Mar 23 23:25:52 piscsi PISCSI[495]: [2024-03-23 23:25:52.111] [info] Attached SCHD 6:0
`

rdmark, 2024-03-24:
"HD1-100MB_HFS.hda" doesn't look like the file name of the Silverlining formatted image. Are the logs from a different attempt? Anyways, there's nothing in this log message that gives any hint. Can you please make sure you enabled trace level logging?

Where did you get the piscsi board from? You might want to double check with them that the board was tested before it was shipped.

Jeremiahdmass, 2024-03-24:
That isn't the image from that issue ticket, but I have also tried it and gotten nothing. The "HD1-100MB_HFS.hda" image was from somewhere random but it should also be Silverlining formatted.  

How do I enable trace-level logging? I may just be an idiot but I looked through the wiki and couldn't find it.

I bought a kit from eBay with just the SMD parts soldered and did the rest myself, so it wasn't tested. I have checked it multiple times and my solder work is good.

rdmark, 2024-03-26:
Look at that, it’s a segmentation fault at the end of the logs. Would it be possible to get a stack trace from that segfault?

Jeremiahdmass, 2024-03-26:
Sorry, I don't really know what that means. I'm not good at this kind of stuff.

Jeremiahdmass, 2024-04-01:
Ok, I have that done. Now what do I need to do?

rdmark, 2024-04-05:
@Jeremiahdmass My apologies, I realized you need to do a `git pull` *after* `git checkout develop` to get the latest code.

The next step would be to run piscsi in the `gdb` debugger, e.g. `sudo gdb --arge piscsi -ID0 /path/to/harddrive.hda`

Then you attempt to boot your system. Once the piscsi process crashes with the segfault, type "bt" in the debugger to get the backtrace. Then share the backtrace here.

Jeremiahdmass, 2024-04-08:
I have fixed that and gotten the latest code, but when I run piscsi with the debugger it doesn't work. I don't think its attaching the disk image, because the activity indicator isn't doing anything. When I run the service normally, the light will blink, but only if an image is mounted.

rdmark, 2024-04-08:
Did you execute the “run” command in the debugger?

Jeremiahdmass, 2024-04-08:
I didn't know that needed to be done🤦‍♂️.

Thread 1 "piscsi" received signal SIGSEGV, Segmentation fault.
0x00000000004dd130 in GPIOBUS::CommandHandShake(std::vector<unsigned char, std::allocator<unsigned char> >&) ()
(gdb) bt
#0  0x00000000004dd130 in GPIOBUS::CommandHandShake(std::vector<unsigned char, std::allocator<unsigned char> >&) ()
#1  0x00000000004a16bc in ScsiController::Command() ()
#2  0x00000000004a15a8 in ScsiController::Selection() ()
#3  0x00000000004a7db0 in PhaseHandler::ProcessPhase() const ()
#4  0x00000000004a11f0 in ScsiController::Process(int) ()
#5  0x00000000004aaec8 in AbstractController::ProcessOnController(int) ()
#6  0x00000000004a916c in ControllerManager::ProcessOnController(int) const ()
#7  0x00000000004836c8 in Piscsi::Process() ()
#8  0x0000000000483030 in Piscsi::run(std::span<char*, 18446744073709551615ul>) ()
#9  0x00000000004e2d58 in main ()
(gdb)

Jeremiahdmass, 2024-04-13:
When I boot the piscsi after the Mac and then try to use the hdsc app the service doesn't crash but the logs are different. I don't know if that means anything but here they are. 

[2024-04-14 00:06:10.923] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
[2024-04-14 00:06:10.923] [trace] (ID 6) - Selection phase
[2024-04-14 00:06:10.924] [trace] (ID 6) - Selection completed
[2024-04-14 00:06:10.924] [trace] (ID 6) - Command phase
[2024-04-14 00:06:14.923] [trace] (ID 6) - Received unknown command: $0
[2024-04-14 00:06:14.923] [debug] (ID 6) - Error status: Sense Key $05, ASC $20
[2024-04-14 00:06:14.924] [trace] (ID 6) - Status phase, status is $02
[2024-04-14 00:06:14.924] [trace] (ID 6) - Sending data, offset: 0, length: 1
[2024-04-14 00:06:22.923] [trace] (ID 6) - Bus Free phase

rdmark, 2024-05-01:
Very nice stacktrace, thank you! The piscsi process seems to be doing absolutely normal things there, and fails at a very early stage when accessing the RPi's GPIO bus. Not to try and lay the blame on your setup, but I wonder if your RPi could be faulty, or if you could have poor connectivity on one or more of the GPIO pins? Have you used all the GPIO pins with other RPi hats, by any chance? Do you have more RPis lying around that you could try?

Jeremiahdmass, 2024-05-02:
I tried a different RPi, and I got the same results. I used the SD card from the other Pi I was using and used jumpers to connect the piscsi board to the GPIO. I triple-checked all the jumper connections before testing and ensured they were all good.

rdmark, 2024-05-03:
Thanks for the thorough tests. 

I released a new stable version of the software earlier this week. May I ask you to upgrade and see if this changes anything? If it looks the same, we may have to start trying older software versions and see if there was an earlier breakage.

Jeremiahdmass, 2024-05-04:
I updated it and I'm still getting the exact same results. What do I need to do to install the older versions?

rdmark, 2024-05-06:
Do you have spare micro SD cards? We have pre-made images for all release versions that you can flash and have a working setup with an old version with little effort.

Jeremiahdmass, 2024-05-06:
I installed the February 2022 version (v22.02.02-lite) and still, nothing.

I got the trace logs and it looks like a very similar error to what I was getting on the latest version.

"[2024-05-06 22:08:53.629] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=7 (with device)
[2024-05-06 22:08:53.629] [trace] virtual void SASIDEV::Command() Command Phase
[2024-05-06 22:08:54.351] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=7 (with device)
[2024-05-06 22:08:54.351] [trace] virtual void SASIDEV::Command() Command Phase
[2024-05-06 22:08:55.629] [trace] virtual void SCSIDEV::Error(ERROR_CODES::sense_key, ERROR_CODES::asc) Error (to status phase)
[2024-05-06 22:08:55.629] [trace] void SASIDEV::Status() Status phase
[2024-05-06 22:08:55.629] [trace] void SASIDEV::Status() Status Phase $02
[2024-05-06 22:08:55.629] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
"