PiSCSI/piscsi #1470: PISCSI fails with "Can't connect to piscsi on host 'localhost', port 6868: Connection refused"

mkegruber, 2024-06-24:
# Info

- Which version of Pi are you using: Raspberry Pi 2 Model B Rev 1.1
- Which github revision of software: 24.04.01
- Which board version: 2.7
- Which computer is the PiSCSI connected to: Amiga 2000
- Which OS you are using (output of 'lsb_release -a'): Raspbian Release 11

# Describe the issue

I completed
- drive setup (https://github.com/PiSCSI/piscsi/wiki/Drive-Setup)
- checked SCSI IDs
- created and attached 2x hda files
- got the Amiga 2000 to detect the drives and formatted them.

After formatting, I shutdown the PI and Amiga.

Then...
1 - rebooted PI
2 - edited piscsi.service, change to

**[Unit]
Description=PISCSI device
After=network.target

[Service]
Type=simple
Restart=always
ExecStart=/usr/local/bin/piscsi -ID1 /home/rascsi/images/WB500MB.hda -ID2 /home/rascsi/images/WORK500MB.hda
ExecStop=/usr/local/bin/scsictl -X
SyslogIdentifier=PISCSI

[Install]
WantedBy=multi-user.target**

3 - rebooted the PI
4 - display during reboot

[FAILED] Failed to start PISCSI service

5 - sudo systemctl status piscsi.service

piscsi.service: Start request repeated too quickly (is this becaue I tried to attach 2x devices on the same command line?)

6 - when I tried scsictl -l after the reboot I get "Error: Can't connect to piscsi on host 'localhost', port 6868: Connection refused", so I created boot/ssh blank file and rebooted, but that did not seem to work.

See screen shot attached.

  
[image]
[image]

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -

stinkerton18, 2024-06-25:
Also, the "Error: Can't connect to piscsi on host 'localhost', port 6868: Connection refused" is normal/expected when the piscsi service itself is fully stopped.

stinkerton18, 2024-07-11:
Hello @mkegruber, just checking in to see if you had read/seen my previous messages and if you still cannot get the PiSCSI service to start after checking/validating access to the image files.

closed 2025-11-07 by rdmark