PiSCSI/piscsi #1474: Can’t boot Solaris 2.5.1 install CD on SPARCstation 5

delan, 2024-07-19:
# Info

- Which version of Pi are you using: Raspberry Pi 4 Model B (1GB)
- Which github revision of software: v24.04.01
- Which board version: akuker 2.6d
- Which computer is the PiSCSI connected to: Sun SPARCstation 5
- Which OS you are using (output of 'lsb_release -a'): official image [2024-04-30-PiSCSI-v24.04.01-arm64-lite.zip](https://github.com/PiSCSI/piscsi/releases/download/v24.04.01/2024-04-30-PiSCSI-v24.04.01-arm64-lite.zip)

# Describe the issue

With the correct CDROM sector size selected for a SPARCstation 5 (#1473), the Solaris 2.5.1 installer kernel now boots without any SCSI errors nor hanging the Pi, but then it fails to mount the root filesystem.

Full logs: [journalctl.txt](https://github.com/user-attachments/files/16313493/journalctl.txt)

[Toshiba XM-3401TA](https://github.com/PiSCSI/piscsi/blob/v24.04.01/python/web/src/drive_properties.json#L374-L385):

[image]
[image]

[Generic CD-ROM block size 512](https://github.com/PiSCSI/piscsi/blob/v24.04.01/python/web/src/drive_properties.json#L434-L445):

[image]
[image]

I think my hardware and termination are ok, because the same setup works fine on a ZuluSCSI. Maybe I made a mistake when assembling my PiSCSI, or damaged it somehow? Any advice on how I can troubleshoot this would be much appreciated.

rdmark, 2024-07-27:
@delan Have you tried to configure different INQUIRE vendor strings? IIRC, many UNIX workstations will only boot from CD-ROM when they encounter devices that identify themselves as a particular bespoke drive model.

We have defined some known-to-work examples in https://github.com/PiSCSI/piscsi/blob/develop/python/web/src/drive_properties.json

rdmark, 2024-07-30:
@delan Were you able to try these device profiles? The Web UI has a handy feature where you can attach a CD-ROM drive using one of those profiles, and then subsequent use of that attached device will inherit those settings that improve compatibility.

delan, 2024-07-30:
Thanks for the advice! I haven’t had time to try it yet sorry, hopefully that will change at some point next month.

rdmark, 2026-09-04:
@delan piscsi software v26.8.1 contains an important bugfix for CD-ROM which may make a difference for your case

delan, 2026-09-06:
thanks! unfortunately i can no longer test this, but thank you for fixing it. feel free to close this for now :)

rdmark, 2026-09-06:
as you wish -- @nsafran1217 reported back in 2021 that their SPARCstation 5 was compatible, but piscsi core was essentially rewritten between 2021 and 2024 so there is a chance that compatibility was degraded since then

closed 2026-09-06 by rdmark